nopnop2002/esp-idf-mqtt-broker #11: ESP32-PICO-D4 (rev1) publish not working

Bitbanger1900, 2019-12-07:
Hi,

I try to get this broker running on an ESP32-PICO-D4. Everything is compiling without errors and when i start the ESP32 after uploading i have a listener on port 1883 and 8080 and i can connect with netcat to the listeners.

When i try to publish something to the broker with for example:
mosquitto_pub -h 192.168.4.1 -m "test" -t "test"  nothing happens and after a wile it stops with the message: Error: Unknown error.

When i try to connect with the test-socket.py it ends with the message socket.timeout: timed out.

i tried with and without the modiffication of mongoose.c

any help much appreciated.

esptool.py v2.9-dev
Serial port /dev/ttyUSB0
Connecting.....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, Embedded Flash, Coding Scheme None

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7184
ho 0 tail 12 room 4
load:0x40078000,len:14824
ho 0 tail 12 room 4
load:0x40080400,len:4468
entry 0x400806ec
[0;32mI (78) boot: Chip Revision: 1[0m
[0;32mI (79) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (43) boot: ESP-IDF v4.1-dev-1086-g93a8603c5 2nd stage bootloader[0m
[0;32mI (43) boot: compile time 23:11:54[0m
[0;32mI (43) boot: Enabling RNG early entropy source...[0m
[0;32mI (49) boot: SPI Speed      : 40MHz[0m
[0;32mI (54) boot: SPI Mode       : DIO[0m
[0;32mI (58) boot: SPI Flash Size : 4MB[0m
[0;32mI (62) boot: Partition Table:[0m
[0;32mI (65) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (87) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (95) boot: End of partition table[0m
[0;32mI (99) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1859c ( 99740) map[0m
[0;32mI (151) esp_image: segment 1: paddr=0x000285c4 vaddr=0x3ffb0000 size=0x032b4 ( 12980) load[0m
[0;32mI (156) esp_image: segment 2: paddr=0x0002b880 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (158) esp_image: segment 3: paddr=0x0002bc88 vaddr=0x40080400 size=0x04390 ( 17296) load[0m
[0;32mI (174) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x78430 (492592) map[0m
[0;32mI (352) esp_image: segment 5: paddr=0x000a8458 vaddr=0x40084790 size=0x10830 ( 67632) load[0m
[0;32mI (393) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (393) boot: Disabling RNG early entropy source...[0m
[0;32mI (394) cpu_start: Pro cpu up.[0m
[0;32mI (397) cpu_start: Application information:[0m
[0;32mI (402) cpu_start: Project name:     esp32-mqtt-broker[0m
[0;32mI (408) cpu_start: App version:      62f365a-dirty[0m
[0;32mI (413) cpu_start: Compile time:     Dec  7 2019 23:11:58[0m
[0;32mI (419) cpu_start: ELF file SHA256:  c2c399b2f90713f1...[0m
[0;32mI (425) cpu_start: ESP-IDF:          v4.1-dev-1086-g93a8603c5[0m
[0;32mI (432) cpu_start: Starting app cpu, entry point is 0x40081170[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (442) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (449) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (455) heap_init: At 3FFB91E8 len 00026E18 (155 KiB): DRAM[0m
[0;32mI (461) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (468) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (474) heap_init: At 40094FC0 len 0000B040 (44 KiB): IRAM[0m
[0;32mI (480) cpu_start: Pro cpu start user code[0m
[0;32mI (499) spi_flash: detected chip: gd[0m
[0;32mI (499) spi_flash: flash io: dio[0m
[0;32mI (500) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (568) main: ESP_WIFI_MODE_AP[0m
I (588) wifi: wifi driver task: 3ffc0c90, prio:23, stack:3584, core=0
[0;32mI (588) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (588) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (618) wifi: wifi firmware version: a44d1c6
I (618) wifi: config NVS flash: enabled
I (618) wifi: config nano formating: disabled
I (618) wifi: Init dynamic tx buffer num: 32
I (618) wifi: Init data frame dynamic rx buffer num: 32
I (628) wifi: Init management frame dynamic rx buffer num: 32
I (628) wifi: Init management short buffer num: 32
I (638) wifi: Init static rx buffer size: 1600
I (638) wifi: Init static rx buffer num: 10
I (648) wifi: Init dynamic rx buffer num: 32
[0;32mI (748) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0[0m
I (748) wifi: mode : softAP (d8:a0:1d:40:6b:a1)
I (748) wifi: Total power save buffer number: 16
I (748) wifi: Init max length of beacon: 752/752
I (758) wifi: Init max length of beacon: 752/752
[0;32mI (758) main: wifi_init_softap finished. SSID:esp32 password:esp32-test
[0;32mI (758) main: WIFI_EVENT event_id=12[0m
[0;32mI (778) main: ESP32 is AP MODE[0m
[0;32mI (778) main: IP Address:  192.168.4.1[0m
[0;32mI (778) main: Subnet mask: 255.255.255.0[0m
[0;32mI (788) main: Gateway:     192.168.4.1[0m
[0;32mI (788) web_socket: HTTP server started on 192.168.4.1:8080[0m
[0;32mI (798) web_socket: MQTT broker started on 192.168.4.1:1883[0m
[0;32mI (808) subscriber: MQTT subscriber started on 192.168.4.1:1883[0m
[0;32mI (838) web_socket: MG_EV_MQTT_CONNECT[0m
I (23868) wifi: new:<1,1>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (23868) wifi: station: 84:3a:4b:cb:e7:68 join, AID=1, bgn, 40U
[0;32mI (23898) main: WIFI_EVENT event_id=14[0m
[0;32mI (23898) main: station 84:3a:4b:cb:e7:68 join, AID=1[0m
[0;32mI (23938) esp_netif_lwip: softAP assign IP to station,IP is: 192.168.4.2[0m
[0;32mI (61818) web_socket: MG_EV_MQTT_PINGREQ[0m
[0;32mI (63748) web_socket: MG_EV_WEBSOCKET_HANDSHAKE_DONE[0m
0;32mI (63748) web_socket: Created proxy connection 0x3ffca294[0m
[0;32mI (122818) web_socket: MG_EV_MQTT_PINGREQ[0m
[0;32mI (123798) web_socket: Closing connection 0x3ffbd490[0m
[0;32mI (123808) web_socket: Closing connection 0x3ffca294[0m

nopnop2002, 2020-02-07:
Sorry for the late reply.

This is my log:
```
I (0) cpu_start: App cpu up.
I (463) heap_init: Initializing. RAM available for dynamic allocation:
I (470) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (476) heap_init: At 3FFB91B0 len 00026E50 (155 KiB): DRAM
I (483) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (489) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (495) heap_init: At 400958F0 len 0000A710 (41 KiB): IRAM
I (502) cpu_start: Pro cpu start user code
I (520) spi_flash: detected chip: gd
I (520) spi_flash: flash io: dio
I (521) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (592) main: ESP_WIFI_MODE_STA
I (632) wifi: wifi driver task: 3ffc0d60, prio:23, stack:3584, core=0
I (632) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (632) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (662) wifi: wifi firmware version: 0849c77
I (662) wifi: config NVS flash: enabled
I (662) wifi: config nano formating: disabled
I (662) wifi: Init dynamic tx buffer num: 32
I (662) wifi: Init data frame dynamic rx buffer num: 32
I (672) wifi: Init management frame dynamic rx buffer num: 32
I (682) wifi: Init management short buffer num: 32
I (682) wifi: Init static rx buffer size: 1600
I (682) wifi: Init static rx buffer num: 10
I (692) wifi: Init dynamic rx buffer num: 32
I (792) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (792) wifi: mode : sta (30:ae:a4:24:bc:5c)
I (792) main: wifi_init_sta finished.
I (792) main: connect to ap SSID:aterm-d5a4ee-g password:7530c4872b4f2
I (792) main: WIFI_EVENT event_id=2
I (932) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (932) wifi: state: init -> auth (b0)
I (932) wifi: state: auth -> assoc (0)
I (942) wifi: state: assoc -> run (10)
I (10942) wifi: state: run -> init (cc00)
I (10942) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (10942) main: WIFI_EVENT event_id=5
I (10952) main: retry to connect to the AP
I (10952) main: connect to the AP fail
I (13002) main: WIFI_EVENT event_id=5
I (13002) main: retry to connect to the AP
I (13002) main: connect to the AP fail
I (13122) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (13122) wifi: state: init -> auth (b0)
I (13122) wifi: state: auth -> assoc (0)
I (13142) wifi: state: assoc -> run (10)
I (13252) wifi: connected with aterm-d5a4ee-g, aid = 6, channel 1, BW20, bssid = f8:b7:97:36:de:52
I (13252) wifi: security type: 4, phy: bgn, rssi: -50
I (13262) wifi: pm start, type: 1

I (13262) main: WIFI_EVENT event_id=4
I (13342) wifi: AP's beacon interval = 102400 us, DTIM period = 1
I (14102) esp_netif_handlers: sta ip: 192.168.10.135, mask: 255.255.255.0, gw: 192.168.10.1
I (14102) main: IP_EVENT event_id=0
I (14102) main: got ip:192.168.10.135
I (14112) main: WIFI_CONNECTED_BIT
I (14112) main: Got IP Address.
I (14112) main: ESP32 is STA MODE
I (14122) main: IP Address:  192.168.10.135
I (14122) main: Subnet mask: 255.255.255.0
I (14132) main: Gateway:     192.168.10.1
I (14132) web_socket: HTTP server started on 192.168.10.135:8080
I (14142) web_socket: MQTT broker started on 192.168.10.135:1883
I (14152) subscriber: MQTT subscriber started on 192.168.10.135:1883
I (14172) web_socket: MG_EV_MQTT_CONNECT
I (14182) subscriber: Subscribing to '#'
I (14192) web_socket: MG_EV_MQTT_SUBSCRIBE
I (14202) subscriber: Subscription acknowledged
I (74212) web_socket: MG_EV_MQTT_PINGREQ
```

IP address and Gateway are the same.
Something is wrong.

```
�[0;32mI (778) main: IP Address: 192.168.4.1�[0m
�[0;32mI (778) main: Subnet mask: 255.255.255.0�[0m
�[0;32mI (788) main: Gateway: 192.168.4.1�[0m
```

nopnop2002, 2020-02-07:
Sorry for the late reply.

This is my log:
```
I (0) cpu_start: Starting scheduler on APP CPU.
I (592) main: ESP_WIFI_MODE_AP
I (632) wifi: wifi driver task: 3ffc0d34, prio:23, stack:3584, core=0
I (632) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (632) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (662) wifi: wifi firmware version: 0849c77
I (662) wifi: config NVS flash: enabled
I (662) wifi: config nano formating: disabled
I (662) wifi: Init dynamic tx buffer num: 32
I (662) wifi: Init data frame dynamic rx buffer num: 32
I (672) wifi: Init management frame dynamic rx buffer num: 32
I (682) wifi: Init management short buffer num: 32
I (682) wifi: Init static rx buffer size: 1600
I (682) wifi: Init static rx buffer num: 10
I (692) wifi: Init dynamic rx buffer num: 32
I (1372) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1382) wifi: mode : softAP (30:ae:a4:24:bc:5d)
I (1382) wifi: Total power save buffer number: 16
I (1382) wifi: Init max length of beacon: 752/752
I (1382) wifi: Init max length of beacon: 752/752
I (1392) main: wifi_init_softap finished. SSID:esp32 password:esp32-test
I (1392) main: WIFI_EVENT event_id=12
I (1402) main: ESP32 is AP MODE
I (1402) main: IP Address:  192.168.4.1
I (1412) main: Subnet mask: 255.255.255.0
I (1412) main: Gateway:     192.168.4.1
I (1422) web_socket: HTTP server started on 192.168.4.1:8080
I (1432) web_socket: MQTT broker started on 192.168.4.1:1883
I (1432) subscriber: MQTT subscriber started on 192.168.4.1:1883
I (1462) web_socket: MG_EV_MQTT_CONNECT
I (1472) subscriber: Subscribing to '#'
I (1482) web_socket: MG_EV_MQTT_SUBSCRIBE
I (1492) subscriber: Subscription acknowledged
I (18602) wifi: new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (18602) wifi: station: 00:13:ef:37:12:f0 join, AID=1, bgn, 40U
I (18622) main: WIFI_EVENT event_id=14
I (18622) main: station 00:13:ef:37:12:f0 join, AID=1
I (18692) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
I (61502) web_socket: MG_EV_MQTT_PINGREQ
I (121522) web_socket: MG_EV_MQTT_PINGREQ
I (181542) web_socket: MG_EV_MQTT_PINGREQ
I (241562) web_socket: MG_EV_MQTT_PINGREQ
I (301582) web_socket: MG_EV_MQTT_PINGREQ
I (361602) web_socket: MG_EV_MQTT_PINGREQ
I (421622) web_socket: MG_EV_MQTT_PINGREQ
I (481642) web_socket: MG_EV_MQTT_PINGREQ
I (481732) wifi: new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (481732) wifi: station: 00:13:ef:c3:0b:b7 join, AID=2, bgn, 40U
I (481752) main: WIFI_EVENT event_id=14
I (481752) main: station 00:13:ef:c3:0b:b7 join, AID=2
I (482032) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.3
I (493192) web_socket: MG_EV_WEBSOCKET_HANDSHAKE_DONE
I (493202) web_socket: Created proxy connection 0x3ffca458
I (493222) web_socket: MG_EV_WEBSOCKET_FRAME
I (493222) web_socket: wm->data[0]=[10] command=[01]
I (493222) web_socket: wm->data[1]=24 wm->size=26
I (493222) web_socket: Forwarding 26 bytes
I (493232) : 10 18 00 04 4d 51 54 54 04 02 00 3c 00 0c 63 6c
I (493232) : 69 65 6e 74 2d 73 6f 63 6b 73
I (493252) web_socket: MG_EV_MQTT_CONNECT
I (493262) web_socket: Responding 4 bytes
I (493262) : 20 02 00 00
I (493262) web_socket: index=0 data_len=2
I (493262) : 20 02 00 00
I (493282) web_socket: MG_EV_WEBSOCKET_FRAME
I (493282) web_socket: wm->data[0]=[82] command=[08]
I (493282) web_socket: wm->data[1]=12 wm->size=14
I (493282) web_socket: Forwarding 14 bytes
I (493292) : 82 0c 00 01 00 07 68 6f 75 73 65 2f 23 00
I (493302) web_socket: MG_EV_MQTT_SUBSCRIBE
I (493312) web_socket: Responding 5 bytes
I (493312) : 90 03 00 01 00
I (493312) web_socket: index=0 data_len=3
I (493312) : 90 03 00 01 00
I (496252) web_socket: MG_EV_WEBSOCKET_FRAME
I (496252) web_socket: wm->data[0]=[30] command=[03]
I (496252) web_socket: wm->data[1]=26 wm->size=28
I (496252) web_socket: Forwarding 28 bytes
I (496262) : 30 1a 00 0b 68 6f 75 73 65 2f 62 75 6c 62 31 48
I (496262) : 65 6c 6c 6f 20 57 6f 72 6c 64 20 30
I (496282) web_socket: MG_EV_MQTT_PUBLISH
I (496292) web_socket: Responding 28 bytes
I (496292) : 30 1a 00 0b 68 6f 75 73 65 2f 62 75 6c 62 31 48
I (496292) : 65 6c 6c 6f 20 57 6f 72 6c 64 20 30
I (496292) subscriber: Got incoming message house/bulb1: Hello World 0
I (496302) web_socket: index=0 data_len=26
I (496312) : 30 1a 00 0b 68 6f 75 73 65 2f 62 75 6c 62 31 48
I (496312) : 65 6c 6c 6f 20 57 6f 72 6c 64 20 30
I (500232) web_socket: MG_EV_WEBSOCKET_FRAME
I (500232) web_socket: wm->data[0]=[30] command=[03]
I (500232) web_socket: wm->data[1]=26 wm->size=28
I (500232) web_socket: Forwarding 28 bytes
I (500242) : 30 1a 00 0b 68 6f 75 73 65 2f 62 75 6c 62 31 48
I (500242) : 65 6c 6c 6f 20 57 6f 72 6c 64 20 31
I (500262) web_socket: MG_EV_MQTT_PUBLISH
I (500272) web_socket: Responding 28 bytes
I (500272) : 30 1a 00 0b 68 6f 75 73 65 2f 62 75 6c 62 31 48
I (500272) : 65 6c 6c 6f 20 57 6f 72 6c 64 20 31
I (500282) web_socket: index=0 data_len=26
I (500272) subscriber: Got incoming message house/bulb1: Hello World 1
I (500282) : 30 1a 00 0b 68 6f 75 73 65 2f 62 75 6c 62 31 48
I (500292) : 65 6c 6c 6f 20 57 6f 72 6c 64 20 31
I (502262) web_socket: Closing connection 0x3ffcab9c
I (502272) web_socket: Closing connection 0x3ffca458
I (502292) web_socket: Closing MQTT connection 0x3ffca6a0
```
---

I tested using test-socket.py.

Mosquitto can web-socket, but the mosquitto_pub client doesn't have websockets support I'm afraid.