nopnop2002/esp-idf-mqtt-broker #12: fatal error: mongoose.h: No such file or directory

p-ranav, 2020-09-16:
Hello, thank you for your work. I'm getting the following error trying to build your project. Any idea how I can fix this? 

Thank you.

```console
fatal error: mongoose.h: No such file or directory
```

```console
pranav@DESKTOP-PR4MHLJ:~/dev/esp-idf-mqtt-broker$ idf.py build
Checking Python dependencies...
Python requirements from /home/pranav/esp/esp-idf/requirements.txt are satisfied.
Executing action: all (aliases: build)
Running cmake in directory /home/pranav/dev/esp-idf-mqtt-broker/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DWARN_UNINITIALIZED=1 -DCCACHE_ENABLE=0 /home/pranav/dev/esp-idf-mqtt-broker"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.25.1")
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/pranav/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/pranav/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/pranav/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/pranav/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/pranav/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: cd79d77
-- Building ESP-IDF components for target esp32
-- Found PythonInterp: /home/pranav/.espressif/python_env/idf4.0_py3.8_env/bin/python (found version "3.8.2")
-- Found Perl: /usr/bin/perl (found version "5.30.0")
-- Adding linker script /home/pranav/dev/esp-idf-mqtt-broker/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/pranav/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /home/pranav/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/pranav/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/pranav/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/pranav/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/pranav/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/pranav/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /home/pranav/esp/esp-idf/components/app_trace /home/pranav/esp/esp-idf/components/app_update /home/pranav/esp/esp-idf/components/asio /home/pranav/esp/esp-idf/components/bootloader /home/pranav/esp/esp-idf/components/bootloader_support /home/pranav/esp/esp-idf/components/bt /home/pranav/esp/esp-idf/components/coap /home/pranav/esp/esp-idf/components/console /home/pranav/esp/esp-idf/components/cxx /home/pranav/esp/esp-idf/components/driver /home/pranav/esp/esp-idf/components/efuse /home/pranav/esp/esp-idf/components/esp-tls /home/pranav/esp/esp-idf/components/esp32 /home/pranav/esp/esp-idf/components/esp_adc_cal /home/pranav/esp/esp-idf/components/esp_common /home/pranav/esp/esp-idf/components/esp_eth /home/pranav/esp/esp-idf/components/esp_event /home/pranav/esp/esp-idf/components/esp_gdbstub /home/pranav/esp/esp-idf/components/esp_http_client /home/pranav/esp/esp-idf/components/esp_http_server /home/pranav/esp/esp-idf/components/esp_https_ota /home/pranav/esp/esp-idf/components/esp_https_server /home/pranav/esp/esp-idf/components/esp_local_ctrl /home/pranav/esp/esp-idf/components/esp_ringbuf /home/pranav/esp/esp-idf/components/esp_rom /home/pranav/esp/esp-idf/components/esp_websocket_client /home/pranav/esp/esp-idf/components/esp_wifi /home/pranav/esp/esp-idf/components/espcoredump /home/pranav/esp/esp-idf/components/esptool_py /home/pranav/esp/esp-idf/components/expat /home/pranav/esp/esp-idf/components/fatfs /home/pranav/esp/esp-idf/components/freemodbus /home/pranav/esp/esp-idf/components/freertos /home/pranav/esp/esp-idf/components/heap /home/pranav/esp/esp-idf/components/idf_test /home/pranav/esp/esp-idf/components/jsmn /home/pranav/esp/esp-idf/components/json /home/pranav/esp/esp-idf/components/libsodium /home/pranav/esp/esp-idf/components/log /home/pranav/esp/esp-idf/components/lwip /home/pranav/dev/esp-idf-mqtt-broker/main /home/pranav/esp/esp-idf/components/mbedtls /home/pranav/esp/esp-idf/components/mdns /home/pranav/esp/esp-idf/components/mqtt /home/pranav/esp/esp-idf/components/newlib /home/pranav/esp/esp-idf/components/nghttp /home/pranav/esp/esp-idf/components/nvs_flash /home/pranav/esp/esp-idf/components/openssl /home/pranav/esp/esp-idf/components/partition_table /home/pranav/esp/esp-idf/components/protobuf-c /home/pranav/esp/esp-idf/components/protocomm /home/pranav/esp/esp-idf/components/pthread /home/pranav/esp/esp-idf/components/sdmmc /home/pranav/esp/esp-idf/components/soc /home/pranav/esp/esp-idf/components/spi_flash /home/pranav/esp/esp-idf/components/spiffs /home/pranav/esp/esp-idf/components/tcp_transport /home/pranav/esp/esp-idf/components/tcpip_adapter /home/pranav/esp/esp-idf/components/ulp /home/pranav/esp/esp-idf/components/unity /home/pranav/esp/esp-idf/components/vfs /home/pranav/esp/esp-idf/components/wear_levelling /home/pranav/esp/esp-idf/components/wifi_provisioning /home/pranav/esp/esp-idf/components/wpa_supplicant /home/pranav/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pranav/dev/esp-idf-mqtt-broker/build
Running ninja in directory /home/pranav/dev/esp-idf-mqtt-broker/build
Executing "ninja all"...
[221/830] cd /home/pranav/dev/esp-idf-mqtt-broker...************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[457/830] Performing configure step for 'bootloader'
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.25.1")
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/pranav/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/pranav/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
[... 96 lines cut ...]
[821/830] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
/home/pranav/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -I/home/pranav/esp/esp-idf/components/newlib/platform_include -I/home/pranav/esp/esp-idf/components/freertos/include -I/home/pranav/esp/esp-idf/components/heap/include -I/home/pranav/esp/esp-idf/components/log/include -I/home/pranav/esp/esp-idf/components/soc/esp32/include -I/home/pranav/esp/esp-idf/components/soc/include -I/home/pranav/esp/esp-idf/compone [... cut ...] -D_GNU_SOURCE -DIDF_VER=\"v4.0.1\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/subscriber.c.obj -MF esp-idf/main/CMakeFiles/__idf_main.dir/subscriber.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/subscriber.c.obj   -c ../main/subscriber.c
../main/subscriber.c:22:10: fatal error: mongoose.h: No such file or directory
 #include "mongoose.h"
          ^~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

nopnop2002, 2020-09-16:
Try cmake or make.

closed 2020-09-28 by nopnop2002

vadaliya, 2020-11-23:
I solved this problem... 
Create CMakeLists.txt file in components folder,
in CMakeLists.txt file write these line -> idf_component_register(SRCS "mongoose.c" INCLUDE_DIRS ".")

nopnop2002, 2020-11-23:
@vadaliya  thank you.