nopnop2002/esp-idf-mqtt-broker #17: mDNS issues

nopnop2002, 2020-09-18:
Repeated CONEECT with MDNS hostname makes connection impossible.

I don't know why.

nopnop2002, 2020-12-09:
Here is the solution.

https://github.com/espressif/esp-idf/issues/6190

pkoevesdi, 2021-01-17:
I can confirm:
With `#define MDNS_STRICT_MODE 1` inside esp-idf/components/mdns/private_include/mdns_private.h it works for long (tested for >900 loops), with strict mode disabled, it repeatable stops working after around 120 loops of above script.
`idf.py --version` gives
`ESP-IDF v4.3-dev-2586-g526f68239-dirty`

nopnop2002, 2021-05-17:
I found the following code in ESP-IDF Ver4:
```
#ifndef CONFIG_MDNS_STRICT_MODE
#define MDNS_STRICT_MODE 0
#else
#define MDNS_STRICT_MODE 1
#endif
```

If you set CONFIG_MDNS_STRICT_MODE = y in sdkconfig.default, it becomes MDNS_STRICT_MODE 1.

nopnop2002, 2022-10-09:
If STRICT MODE is specified in ESP-IDF V4.4.1, mDNS name resolution will not be possible after long operation.

Therefore, STRICT MODE was abolished.

```
#
# mDNS
#
#CONFIG_MDNS_STRICT_MODE=y
```

nopnop2002, 2022-10-09:
- esp-idf V4.3 or earlier   
 You will need to manually change the mDNS strict mode according to [this](https://github.com/espressif/esp-idf/issues/6190) instruction.   
- esp-idf V4.4  
 If you set CONFIG_MDNS_STRICT_MODE = y in sdkconfig.defaults, the firmware will be built with MDNS_STRICT_MODE.   
 __If MDNS_STRICT_MODE is not set, mDNS name resolution will not be possible after long-term operation.__   
- esp-idf V4.4.1   
 mDNS component has been updated.   
 If you set CONFIG_MDNS_STRICT_MODE = y in sdkconfig.defaults, the firmware will be built with MDNS_STRICT_MODE.   
 __Even if MDNS_STRICT_MODE is set, mDNS name resolution will not be possible after long-term operation.__   
- esp-idf V5.0 or later   
 mDNS component has been updated.   
 Long-term operation is possible without setting MDNS_STRICT_MODE.   
 The following lines in sdkconfig.defaults should be removed before menuconfig.   
 ```CONFIG_MDNS_STRICT_MODE=y```

nopnop2002, 2024-09-18:
With ESP-IDF V5, CONFIG_MDNS_STRICT_MODE has been deprecated.

closed 2024-09-18 by nopnop2002