nopnop2002/esp-idf-mqtt-broker #19: error building

pkoevesdi, 2021-01-17:
I freshly installed esp-idf on ubuntu 20.04.1 after https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started and prepared esp-idf-mqtt-broker after README.md, each with minute care.
In `menuconfig` I changed nothing but the wifi ssid and password.
While execution of the last command `make flash`, after lot of seemingly succesful builds it breaks with the follwing output:
```
CC build/main/web_socket.o
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c: In function 'unproxy':
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:36:58: error: 'struct mg_connection' has no member named 'user_data'; did you mean 'fn_data'?
   struct mg_connection *pc = (struct mg_connection *) c->user_data;
                                                          ^~~~~~~~~
                                                          fn_data
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:38:7: error: 'struct mg_connection' has no member named 'flags'
     pc->flags |= MG_F_CLOSE_IMMEDIATELY;
       ^~
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:38:18: error: 'MG_F_CLOSE_IMMEDIATELY' undeclared (first use in this function)
     pc->flags |= MG_F_CLOSE_IMMEDIATELY;
                  ^~~~~~~~~~~~~~~~~~~~~~
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:38:18: note: each undeclared identifier is reported only once for each function it appears in
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:39:9: error: 'struct mg_connection' has no member named 'user_data'; did you mean 'fn_data'?
     pc->user_data = NULL;
         ^~~~~~~~~
         fn_data
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:40:8: error: 'struct mg_connection' has no member named 'user_data'; did you mean 'fn_data'?
     c->user_data = NULL;
        ^~~~~~~~~
        fn_data
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c: In function 'proxy_handler':
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:53:10: error: 'MG_EV_RECV' undeclared (first use in this function); did you mean 'MG_EV_READ'?
     case MG_EV_RECV: {
          ^~~~~~~~~~
          MG_EV_READ
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:54:62: error: 'struct mg_connection' has no member named 'user_data'; did you mean 'fn_data'?
       struct mg_connection *pc = (struct mg_connection *) c->user_data;
                                                              ^~~~~~~~~
                                                              fn_data
In file included from /home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:17:
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:56:53: error: 'struct mg_connection' has no member named 'recv_mbuf'
         ESP_LOGI(TAG, "Responding %d bytes", (int) c->recv_mbuf.len);
                                                     ^~
/home/user/esp/esp-idf/components/log/include/esp_log.h:334:137: note: in definition of macro 'ESP_LOG_LEVEL'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
/home/user/esp/esp-idf/components/log/include/esp_log.h:301:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:56:9: note: in expansion of macro 'ESP_LOGI'
         ESP_LOGI(TAG, "Responding %d bytes", (int) c->recv_mbuf.len);
         ^~~~~~~~
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:56:53: error: 'struct mg_connection' has no member named 'recv_mbuf'
         ESP_LOGI(TAG, "Responding %d bytes", (int) c->recv_mbuf.len);
                                                     ^~
/home/user/esp/esp-idf/components/log/include/esp_log.h:335:137: note: in definition of macro 'ESP_LOG_LEVEL'
         else if (level==ESP_LOG_WARN )      { esp_log_write(ESP_LOG_WARN,       tag, LOG_FORMAT(W, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
/home/user/esp/esp-idf/components/log/include/esp_log.h:301:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:56:9: note: in expansion of macro 'ESP_LOGI'
         ESP_LOGI(TAG, "Responding %d bytes", (int) c->recv_mbuf.len);
         ^~~~~~~~
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:56:53: error: 'struct mg_connection' has no member named 'recv_mbuf'
         ESP_LOGI(TAG, "Responding %d bytes", (int) c->recv_mbuf.len);
                                                     ^~
/home/user/esp/esp-idf/components/log/include/esp_log.h:336:137: note: in definition of macro 'ESP_LOG_LEVEL'
         else if (level==ESP_LOG_DEBUG )     { esp_log_write(ESP_LOG_DEBUG,      tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
/home/user/esp/esp-idf/components/log/include/esp_log.h:301:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGI( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_INFO,    tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:56:9: note: in expansion of macro 'ESP_LOGI'
         ESP_LOGI(TAG, "Responding %d bytes", (int) c->recv_mbuf.len);
[... 159 lines cut ...]
       ^~
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:189:5: error: implicit declaration of function 'mg_set_protocol_mqtt' [-Werror=implicit-function-declaration]
     mg_set_protocol_mqtt(nc);
     ^~~~~~~~~~~~~~~~~~~~
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:177:27: warning: unused variable 'brk' [-Wunused-variable]
     struct mg_mqtt_broker brk;
                           ^~~
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c: In function 'http_server':
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:214:5: error: too many arguments to function 'mg_mgr_init'
     mg_mgr_init(&mgr, NULL);
     ^~~~~~~~~~~
In file included from /home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:23:
/home/user/esp/esp-idf-mqtt-broker/components/mongoose/mongoose.h:640:6: note: declared here
 void mg_mgr_init(struct mg_mgr *);
      ^~~~~~~~~~~
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:216:8: warning: assignment to 'struct mg_connection *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
     nc = mg_bind(&mgr, server_address, http_handler);
        ^
/home/user/esp/esp-idf-mqtt-broker/main/web_socket.c:222:5: error: implicit declaration of function 'mg_set_protocol_http_websocket' [-Werror=implicit-function-declaration]
     mg_set_protocol_http_websocket(nc);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [/home/user/esp/esp-idf/make/component_wrapper.mk:292: web_socket.o] Fehler 1
make: *** [/home/user/esp/esp-idf/make/project.mk:635: component-main-build] Fehler 2
```

pkoevesdi, 2021-01-17:
Ok, I went back to the last mongoose commit before version 7.0: 34d4fb8cf03cdd74773f594aaa1502811841f195
Now this issue is gone.
Please update README.md.

Michelebuffa, 2021-01-19:
Makefile:8: /make/project.mk: File o directory non esistente
make: ***  Nessuna regola per generare l'obiettivo «/make/project.mk».  Arresto.

nopnop2002, 2021-03-06:
Try latest.
The latest mongoose library is available.