nopnop2002/esp-idf-mqtt-broker #20: Error linking `mongoose.h`

pkoevesdi, 2021-01-24:
Now, that I have managed to build this project and get it to work, I face the following problem:
From the command line inside the project folder, `make flash` works fine, but `idf.py flash` doesn't: the compiler doesn't find `mongoose.h`.
So, probably some misconfiguration concerning the folders to include?
Why do I want to make it work with `idf.py flash`? Because, that's the default command inside the ESP-IDF-Extension of VSCode, which get's executed when pressing the button `ESP-IDF Build, Flash and Monitor`.

What is the cleanest, most future-proof way of making it work with `idf.py flash`?

pkoevesdi, 2021-01-24:
Ok, found out, the file `CMakeLists.txt` must also be created in the mongoose directory.
I made it up with the following content:
```
idf_component_register(SRCS "mongoose.c"
                       INCLUDE_DIRS ".")
```
Please, put this into Readme.

nopnop2002, 2021-03-06:
@pkoevesdi 

Try latest.
The latest mongoose library is available.