nopnop2002/esp-idf-mqtt-broker #28: Trouble while connection ESP as a SoftAP to any device

parthshahec108, 2021-03-31:
Hello,

We have used this example and configured ESP32 as a SoftAP and trying to connect to the device (laptop or mobile).

While performing above example, it has been noted that the connection between ESP32 SoftAP and a client is not stable sometimes i.e. connection takes some time to connect, re-connection happens in-between.

Could you please help us here?

Regards.

nopnop2002, 2021-03-31:
I tested SoftAP mode.

This is client:
```
#!/bin/bash
#set -x
count=1
while :
do
    payload="test${count}"
    mosquitto_pub -h 192.168.4.1 -p 1883 -t "test/test" -m ${payload}
    count=$((++count))
    sleep 1
done
```

Connect to MQTT & Publish & Disconnect.
No problem.

Probably a client-side issue.

nopnop2002, 2021-03-31:
I changed from tcpip_adapter_init to esp_netif_init.
Try latest.

```
#!/bin/bash
#set -x
count=1
while :
do
        #payload=$(date +"%T")
        payload="text${count}"
        echo ${payload}
        mosquitto_pub -h 192.168.4.1 -p 1883 -t "test/test" -m payload
        count=$((++count))
        sleep 1
done
```

nopnop2002, 2021-04-02:
I conducted a continuous endurance test of softap.

As a result, it was found that there is no problem with SoftAP of ESP-IDF.

There is no problem even if the connection and disconnection are repeated 27,000 times or more.

This is a problem on the mongoose library side.

nopnop2002, 2021-04-09:
I found memory leek.
Fixed it.

closed 2022-03-25 by nopnop2002