nopnop2002/esp-idf-mqtt-broker #30: Undefined pcTaskGetName symbol at link

readysteadygo2006, 2021-04-09:
Hi, 
When I build I get an error:
undefined reference to `pcTaskGetName'
I've checked that FreeRTOSConfig.h in the sdk makes it available, it does.

Is there something else I need to do?

nopnop2002, 2021-04-09:
Can you build this code?

```
#include <expat.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_spi_flash.h"
#include "esp_efuse.h"

#include "cJSON.h"

#include "lwip/init.h"

void app_main()
{
    printf("freeRTOS version:%s\n", tskKERNEL_VERSION_NUMBER);
    printf("NEWLIB version:%s\n", _NEWLIB_VERSION);
    printf("portTICK_RATE_MS:%d msec/tick\n", portTICK_RATE_MS);
    printf("lwIP version:%d-%d-%d-%d\n",
      LWIP_VERSION_MAJOR,LWIP_VERSION_MINOR,
      LWIP_VERSION_REVISION,LWIP_VERSION_RC);
    printf("ESP-IDF version:%s\n", esp_get_idf_version());
}
```

readysteadygo2006, 2021-04-10:
Hi, 

My response is:
freeRTOS version:V8.2.0
NEWLIB version:3.0.0
portTICK_RATE_MS:10 msec/tick
lwIP version:2-1-3-0
ESP-IDF version:v4.1-dev-2010-g3b86e7e-dirty

I'll check codebase, to see if some other define is required.

It no success, I'll move to later version.

This is not as important as the subscribe not working, I really only 
highlighted it incase it was the cause of the subscribe problem!

nopnop2002, 2021-04-10:
OK.
Try next.

```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_spi_flash.h"
#include "lwip/init.h"

void app_main()
{
    printf("freeRTOS version:%s\n", tskKERNEL_VERSION_NUMBER);
    printf("NEWLIB version:%s\n", _NEWLIB_VERSION);
    printf("portTICK_RATE_MS:%d msec/tick\n", portTICK_RATE_MS);
    printf("lwIP version:%d-%d-%d-%d\n",
      LWIP_VERSION_MAJOR,LWIP_VERSION_MINOR,
      LWIP_VERSION_REVISION,LWIP_VERSION_RC);
    printf("ESP-IDF version:%s\n", esp_get_idf_version());
    printf("pctaskname=%s\n", pcTaskGetName(NULL));
}
```

The following is mine.
```
freeRTOS version:V10.2.1
NEWLIB version:3.0.0
portTICK_RATE_MS:10 msec/tick
lwIP version:2-1-3-0
ESP-IDF version:v4.3-dev-2586-g526f68239-dirty
pctaskname=main
```

nopnop2002, 2021-04-11:
>This is not as important as the subscribe not working,

No.

This is very important,

readysteadygo2006, 2021-04-11:
Hi,

Updated to latest version of idf. Now get:
freeRTOS version:V10.2.1
NEWLIB version:3.0.0
portTICK_RATE_MS:10 msec/tick
lwIP version:2-1-3-0
ESP-IDF version:v4.3-beta2-2-g9a2d251

Now have no issue with pcTaskName

However still get no response is subscribe to a publish!

Do you have any ideas on why this is?

If I uncomment this section in mqtt_server.c, I get a response:
#if 0
		// Client connects. Add to the will list
		if (willFlag == 1) {
		  struct will *will = calloc(1, sizeof(*will));
		  will->c = c;
		  will->topic = mg_strdup(topic);
		  will->payload = mg_strdup(payload);
		  will->qos = qos;
		  will->retain = retain;
		  LIST_ADD_HEAD(struct will, &s_wills, will);
		  ESP_LOGD(pcTaskGetName(NULL), "WILL ADD %p [%.*s] [%.*s] %d %d", 
			c->fd, (int) will->topic.len, will->topic.ptr, (int) will->payload.len, will->payload.ptr, will->qos, will->retain);
		}
#endif

Are you expecting that I should have to do this?

Cheers

readysteadygo2006, 2021-04-11:
Sorry, this is the section which needs to be added:
#if 0
		// Client connects. Add to the client-id list
		struct client *client = calloc(1, sizeof(*client));
		client->c = c;
		client->cid = mg_strdup(cid);
		LIST_ADD_HEAD(struct client, &s_clients, client);
		ESP_LOGD(pcTaskGetName(NULL), "CLIENT ADD %p [%.*s]", c->fd, (int) client->cid.len, client->cid.ptr);
		ESP_LOGI(pcTaskGetName(NULL), "CLIENT ADD %p", client);
#endif

readysteadygo2006, 2021-04-11:
Sorry wrong again, its this section:
 	  case MQTT_CMD_SUBSCRIBE: {
		// Client subscribe. Add to the subscription list
		ESP_LOGI(pcTaskGetName(NULL), "MQTT_CMD_SUBSCRIBE");
		//_mg_mqtt_dump("SUBSCRIBE", mm);
//#if 0
		int pos = 4;  // Initial topic offset, where ID ends
		uint8_t qos;
		struct mg_str topic;
		while ((pos = mg_mqtt_next_sub(mm, &topic, &qos, pos)) > 0) {
		  struct sub *sub = calloc(1, sizeof(*sub));
		  sub->c = c;
		  sub->topic = mg_strdup(topic);
		  sub->qos = qos;
		  LIST_ADD_HEAD(struct sub, &s_subs, sub);
		  ESP_LOGI(pcTaskGetName(NULL), "SUB ADD %p [%.*s]", c->fd, (int) sub->topic.len, sub->topic.ptr);
		}
//#endif

readysteadygo2006, 2021-04-11:
However when added I notice, that the subscription make duplicate entries in the sublist:

extract from debug below:
I (57618) BROKER: SUB[b] 0x38 [7c:9e:bd:39:1e:6c]
I (57618) BROKER: SUB[b] 0x39 [b8:f0:09:cd:42:18]
I (57628) BROKER: SUB[b] 0x39 [b8:f0:09:cd:42:18]
I (57628) BROKER: SUB[b] 0x39 [b8:f0:09:cd:42:18]
I (57638) BROKER: SUB[b] 0x39 [b8:f0:09:cd:42:18]
I (57638) BROKER: SUB[b] 0x39 [b8:f0:09:cd:42:18]
I (57648) BROKER: SUB[b] 0x39 [b8:f0:09:cd:42:18]
I (57648) BROKER: SUB[b] 0x39 [b8:f0:09:cd:42:18]

Is this work-in-progress?

I've not looked in detail yet, but I would expect that you would need to check for existence at addition, is this already done
within the api?

nopnop2002, 2021-04-11:
Try latest.

readysteadygo2006, 2021-04-11:
Hi,

Tried latest! Now subscriber work!

No longer seem sublist growing with same entry!

I'm using PAHO python client and esp32 native clients:

The PAHO clients seem to work fine, I'm getting a problem with the esp32 client failing 
to send messages. Initially it succeed, but then it fails!

I was playing with the keepalive config of the esp32 and the PAHO clients, but no success.

Will do some systematic changes tomorrow to try to find source of problem.

I you have any thoughts they would be very much appreciated!

nopnop2002, 2021-04-11:
OK
I am waiting for your findings.

readysteadygo2006, 2021-04-17:
Hi, just wanted to give you an update.

Originally was using esp32 idf native mqtt client, but had problems with pub/sub.
Problems may have been due to manner of use (blocking), however choose to switch
to using mongoose clients, as you had shown already working.

Replicated you approach of creating threads for pub/sub, communicated with threads 
from main via freertos  events

Ran pub/sub/broker/appthreads on single device.

All working fine! System ran overnight no problem, with mongoose sub/pub and
PAHO pub/sub from python.

Really good project, thanks!