nopnop2002/esp-idf-mqtt-broker #31: No response to mqtt_sub.py

readysteadygo2006, 2021-04-09:
I've built the system working around pcTaskGetName undefined by just adding:

char *pcTaskGetName(void * param) {
	return "DummyTask";
}

But when I run the test scripts:

mqtt_sub.py and mqtt_pub.py I dont get any output.

If I look at the log from the esp32, I can see the published timestamps
I (6837) DummyTask: PUB 0x3b [2021/04/09-12:59:14] -> [test]
I (6847) DummyTask: MG_EV_CLOSE 0x3b
I (7377) DummyTask: PUBSLISHED TickCount=678 -> esp32
I (7377) DummyTask: PUB 0x3a [TickCount=678] -> [esp32]
I (8037) DummyTask: CONNECT

Any help would be appreciated!

readysteadygo2006, 2021-04-09:
Hi,

Noticed in mqtt_server.c

Following code is commented out:
//#if 0
		int pos = 4;  // Initial topic offset, where ID ends
		uint8_t qos;
		struct mg_str topic;
		while ((pos = mg_mqtt_next_sub(mm, &topic, &qos, pos)) > 0) {
		  struct sub *sub = calloc(1, sizeof(*sub));
		  sub->c = c;
		  sub->topic = mg_strdup(topic);
		  sub->qos = qos;
		  LIST_ADD_HEAD(struct sub, &s_subs, sub);
		  ESP_LOGI(pcTaskGetName(NULL), "SUB ADD %p [%.*s]", c->fd, (int) sub->topic.len, sub->topic.ptr);
		}
//#endif

When uncommented, now get responses to subscribe.

Are you experimenting in some way, is any of the other code in #if 0 blocks needed?

nopnop2002, 2022-07-01:
pcTaskGetName is a function provided by freeRTOS.

closed 2022-07-01 by nopnop2002