nopnop2002/esp-idf-mqtt-broker #32: How can i use my server certificates fot tls in MQTT Broker

Darsh-Dev, 2021-04-15:
Hi,

Are there any steps that help me to use my server certificate during the TLS handshake in the MQTT broker example?

Thanks

nopnop2002, 2021-04-15:
I don't know.

There is some code about TLS.

```
cd esp-idf-mqtt-broker/components/mongoose

$ grep -rn TLS *
Makefile:5:SSL ?= MBEDTLS
Makefile:19:ifeq "$(SSL)" "MBEDTLS"
Makefile:20:MBEDTLS_DIR ?= $(shell brew --cellar mbedtls)
Makefile:21:MBEDTLS_VER ?= $(shell brew info mbedtls --json | jq -j .[0].installed[0].version)
Makefile:22:MBEDTLS ?= $(MBEDTLS_DIR)/$(MBEDTLS_VER)
Makefile:23:CFLAGS += -DMG_ENABLE_MBEDTLS=1 -I$(MBEDTLS)/include -I/usr/include
Makefile:24:LDFLAGS ?= -L$(MBEDTLS)/lib -lmbedtls -lmbedcrypto -lmbedx509
Makefile:69:    infer run -- cc test/unit_test.c -c -W -Wall -Werror -Isrc -I. -O2 -DMG_ENABLE_MBEDTLS=1 -DMG_ENABLE_LINES -I/usr/local/Cellar/mbedtls/2.23.0/include  -DMG_ENABLE_IPV6=1 -g -o /dev/null
README.md:80:* SSL/TLS support: mbedTLS, OpenSSL or custom (via API)
docs/README.md:20:   - plain TCP, plain UDP, SSL/TLS (over TCP, one-way or two-way)
docs/README.md:177:  unsigned is_tls : 1;         // TLS-enabled connection
docs/README.md:178:  unsigned is_tls_hs : 1;      // TLS handshake is in progress
docs/README.md:219:|`MG_ENABLE_MBEDTLS` | 0 | Enable Mbed TLS library |
docs/README.md:373:  void *tls;                   // TLS specific data
docs/README.md:379:  unsigned is_tls : 1;         // TLS-enabled connection
docs/README.md:380:  unsigned is_tls_hs : 1;      // TLS handshake is in progress
docs/README.md:1007:## TLS
docs/README.md:1022:Initialise TLS on a given connection.
examples/http-proxy-client/Makefile:3:MBEDTLS_DIR ?=
examples/http-proxy-client/Makefile:5:ifeq "$(MBEDTLS_DIR)" ""
examples/http-proxy-client/Makefile:7:CFLAGS += -DMG_ENABLE_MBEDTLS=1 -I$(MBEDTLS_DIR)/include -I/usr/include
examples/http-proxy-client/Makefile:8:CFLAGS += -L$(MBEDTLS_DIR)/lib -lmbedtls -lmbedcrypto -lmbedx509
examples/http-proxy-client/main.c:8:// To enable SSL/TLS for this client, build it like this:
examples/http-proxy-client/main.c:9://    make MBEDTLS_DIR=/path/to/your/mbedtls/installation
examples/http-proxy-client/main.c:28:    // If target URL is SSL/TLS, command client connection to use TLS
examples/http-proxy-client/ca.pem:46:ksLi4xaNmjICq44Y3ekQEe5+NauQrz4wlHrQMz2nZQ/1/I6eYs9HRCwBXbsdtTLS
examples/http-streaming-client/Makefile:4:ifneq ($(MBEDTLS),)
examples/http-streaming-client/Makefile:5:CFLAGS += -DMG_ENABLE_MBEDTLS=1 -I$(MBEDTLS)/include -I/usr/include
examples/http-streaming-client/Makefile:6:CFLAGS += -L$(MBEDTLS)/lib -lmbedtls -lmbedcrypto -lmbedx509
examples/http-streaming-client/main.c:12:// To enable SSL/TLS for this client, build it like this:
examples/http-streaming-client/main.c:13://    make MBEDTLS=/path/to/your/mbedtls/installation
examples/http-streaming-client/main.c:27:    // If s_url is https://, tell client connection to use TLS
examples/http-streaming-client/ca.pem:46:ksLi4xaNmjICq44Y3ekQEe5+NauQrz4wlHrQMz2nZQ/1/I6eYs9HRCwBXbsdtTLS
examples/socks5-server/Makefile:3:MBEDTLS_DIR ?=
examples/socks5-server/Makefile:5:ifeq "$(MBEDTLS_DIR)" ""
examples/socks5-server/Makefile:7:CFLAGS += -DMG_ENABLE_MBEDTLS=1 -I$(MBEDTLS_DIR)/include -I/usr/include
examples/socks5-server/Makefile:8:CFLAGS += -L$(MBEDTLS_DIR)/lib -lmbedtls -lmbedcrypto -lmbedx509
examples/http-reverse-proxy/Makefile:3:MBEDTLS_DIR ?=
examples/http-reverse-proxy/Makefile:5:ifeq "$(MBEDTLS_DIR)" ""
examples/http-reverse-proxy/Makefile:7:CFLAGS += -DMG_ENABLE_MBEDTLS=1 -I$(MBEDTLS_DIR)/include -I/usr/include
examples/http-reverse-proxy/Makefile:8:CFLAGS += -L$(MBEDTLS_DIR)/lib -lmbedtls -lmbedcrypto -lmbedx509
examples/http-reverse-proxy/main.c:8:// To enable SSL/TLS, build it like this:
examples/http-reverse-proxy/main.c:9://    make MBEDTLS_DIR=/path/to/your/mbedtls/installation
examples/http-reverse-proxy/ca.pem:46:ksLi4xaNmjICq44Y3ekQEe5+NauQrz4wlHrQMz2nZQ/1/I6eYs9HRCwBXbsdtTLS
examples/esp8266/src/main/main.c:20:// To enable TLS for HTTP,
examples/esp8266/src/main/main.c:22://   2. Add TLS init snippet for the connection, see examples/http-client
examples/mqtt-client/Makefile:2:MBEDTLS_DIR ?=
examples/mqtt-client/Makefile:4:ifeq "$(MBEDTLS_DIR)" ""
examples/mqtt-client/Makefile:6:CFLAGS += -DMG_ENABLE_MBEDTLS=1 -I$(MBEDTLS_DIR)/include -I/usr/include
examples/mqtt-client/Makefile:7:CFLAGS += -L$(MBEDTLS_DIR)/lib -lmbedtls -lmbedcrypto -lmbedx509
examples/mqtt-client/main.c:10:// To enable SSL/TLS for this client, build it like this:
examples/mqtt-client/main.c:11://    make MBEDTLS_DIR=/path/to/your/mbedtls/installation
examples/mqtt-client/main.c:23:    // If target URL is SSL/TLS, command client connection to use TLS
examples/mqtt-client/ca.pem:46:ksLi4xaNmjICq44Y3ekQEe5+NauQrz4wlHrQMz2nZQ/1/I6eYs9HRCwBXbsdtTLS
examples/http-restful-server/Makefile:3:ifeq "$(MBEDTLS_DIR)" ""
examples/http-restful-server/Makefile:5:CFLAGS += -DMG_ENABLE_MBEDTLS=1 -I$(MBEDTLS_DIR)/include -I/usr/include
examples/http-restful-server/Makefile:6:CFLAGS += -L$(MBEDTLS_DIR)/lib -lmbedtls -lmbedcrypto -lmbedx509
examples/http-restful-server/main.c:10:// To enable SSL/TLS (using self-signed certificates in PEM files),
examples/http-restful-server/main.c:12://    2. make MBEDTLS_DIR=/path/to/your/mbedtls/installation
examples/http-restful-server/main.c:22:    // If s_listen_on URL is https://, tell listening connection to use TLS
examples/http-restful-server/server.pem:29:                TLS Web Server Authentication
examples/websocket-client/ca.pem:46:ksLi4xaNmjICq44Y3ekQEe5+NauQrz4wlHrQMz2nZQ/1/I6eYs9HRCwBXbsdtTLS
[... 42 lines cut ...]
src/tls.c:45:    mg_error(c, "TLS handshake: -%#x", -rc);  // Error
[... cut ...]
src/sock.c:530:    // TLS might have stuff buffered, so dig everything
src/config.h:11:#ifndef MG_ENABLE_MBEDTLS
src/config.h:12:#define MG_ENABLE_MBEDTLS 0
test/data/ss_client.pem:29:                TLS Web Client Authentication
test/data/ss_server.pem:29:                TLS Web Server Authentication
test/data/ca.pem:46:ksLi4xaNmjICq44Y3ekQEe5+NauQrz4wlHrQMz2nZQ/1/I6eYs9HRCwBXbsdtTLS
test/unit_test.c:613:#if MG_ENABLE_MBEDTLS || MG_ENABLE_OPENSSL
test/unit_test.c:664:#if MG_ENABLE_MBEDTLS || MG_ENABLE_OPENSSL
```

Darsh-Dev, 2021-04-16:
Hi @nopnop2002,

Thanks for the update. I want to know that is TSL feature is used in the MQTT broker/server code ([mqtt_server.c](https://github.com/nopnop2002/esp-idf-mqtt-broker/blob/master/main/mqtt_server.c)) ? If yes then how can I enable and use my server certificate for a secure connection?

Thanks

nopnop2002, 2021-04-16:
This code may be helpful.

https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/esp_http_client_example.c#L29

Darsh-Dev, 2021-04-22:
Hi @nopnop2002,

Issue resolve please find below patch to enable the mqtt_broker with tls.

>  **diff --git a/mongoose.c b/mongoose.c**
static void accept_conn(struct mg_mgr *mgr, struct mg_connection *lsn) {
c->pfn_data = lsn->pfn_data;
c->fn = lsn->fn;
c->fn_data = lsn->fn_data;
+#if MG_ENABLE_MBEDTLS
+struct mg_tls_opts opts = {.ca = NULL,
+.cert = srv_cert,	/* TLS server certificate */
+.certkey = srv_key};/* TLS server private key */
+mg_tls_init(c, &opts);
+#endif
**diff --git a/mongoose.h b/mongoose.h**
 #ifndef MG_ENABLE_MBEDTLS
-#define MG_ENABLE_MBEDTLS 0
+#define MG_ENABLE_MBEDTLS 1

nopnop2002, 2021-10-12:
@stefanobignotti 

Do you specify this?

https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/CMakeLists.txt#L6

stefanobignotti, 2021-10-13:
@nopnop2002
I solved my problem. It was a problem in broker poll delay, now it's working.
Thanks