nopnop2002/esp-idf-mqtt-broker #33: Building help

lucagir, 2021-04-22:
Hello,
I am writing to you because for me it is the first approach with IDF and therefore I could have something wrong.

I am very interested in this project to allow me to bring a mqtt server to my ESP32 board.

I followed the guide provided with the project step by step but at the moment of the last step everything crashes with a compilation error.

Can I ask you for help to understand what I did wrong in the build procedure?

`[1/11] Performing build step for 'bootloader'
ninja: no work to do.
[2/9] Linking CXX static library esp-idf\mongoose\libmongoose.a
[3/9] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/mqtt_server.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/mqtt_server.c.obj 
ccache C:\Users\user\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -IC:/Users/user/Desktop/esp-idf/components/newlib/platform_include -IC:/Users/user/Desktop/esp-idf/components/freertos/include -IC:/Users/user/Desktop/esp-idf/components/freertos/xtensa/include -IC:/Users/user/Desktop/esp-idf/components/heap/include -IC:/Users/user/Desktop/esp-idf/components/log/include -IC:/Users/user/Desktop/esp-idf/components/lwip/include/apps -IC:/Users/user/Desktop/esp-idf/components/lwip/include/apps/sntp -IC:/Users/user/Desktop/esp-idf/components/lwip/lwip/src/include -IC:/Users/user/Desktop/esp-idf/components/lwip/port/esp32/include -IC:/Users/user/Desktop/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/user/Desktop/esp-idf/components/soc/src/esp32/. -IC:/Users/user/Desktop/esp-idf/components/soc/src/esp32/include -IC:/Users/user/Desktop/esp-idf/components/soc/include -IC:/Users/user/Desktop/esp-idf/components/esp_rom/include -IC:/Users/user/Desktop/esp-idf/components/esp_common/include -IC:/Users/user/Desktop/esp-idf/components/esp_system/include -IC:/Users/user/Desktop/esp-idf/components/xtensa/include -IC:/Users/user/Desktop/esp-idf/components/xtensa/esp32/include -IC:/Users/user/Desktop/esp-idf/components/esp32/include -IC:/Users/user/Desktop/esp-idf/components/driver/include -IC:/Users/user/Desktop/esp-idf/components/driver/esp32/include -IC:/Users/user/Desktop/esp-idf/components/esp_ringbuf/include -IC:/Users/user/Desktop/esp-idf/components/efuse/include -IC:/Users/user/Desktop/esp-idf/components/efuse/esp32/include -IC:/Users/user/Desktop/esp-idf/components/espcoredump/include -IC:/Users/user/Desktop/esp-idf/components/esp_timer/include -IC:/Users/user/Desktop/esp-idf/components/esp_ipc/include -IC:/Users/user/Desktop/esp-idf/components/soc/soc/esp32/include -IC:/Users/user/Desktop/esp-idf/components/soc/soc/esp32/../include -IC:/Users/user/Desktop/esp-idf/components/soc/soc/esp32/private_include -IC:/Users/user/Desktop/esp-idf/components/vfs/include -IC:/Users/user/Desktop/esp-idf/components/esp_wifi/include -IC:/Users/user/Desktop/esp-idf/components/esp_wifi/esp32/include -IC:/Users/user/Desktop/esp-idf/components/esp_event/include -IC:/Users/user/Desktop/esp-idf/components/esp_netif/include -IC:/Users/user/Desktop/esp-idf/components/esp_eth/include -IC:/Users/user/Desktop/esp-idf/components/tcpip_adapter/include -IC:/Users/user/Desktop/esp-idf/components/app_trace/include -IC:/Users/user/Desktop/esp-idf/components/mbedtls/port/include -IC:/Users/user/Desktop/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/user/Desktop/esp-idf/components/mbedtls/esp_crt_bundle/include -IC:/Users/user/Desktop/esp-idf/components/bootloader_support/include -IC:/Users/user/Desktop/esp-idf/components/app_update/include -IC:/Users/user/Desktop/esp-idf/components/spi_flash/include -IC:/Users/user/Desktop/esp-idf/components/wpa_supplicant/include -IC:/Users/user/Desktop/esp-idf/components/wpa_supplicant/port/include -IC:/Users/user/Desktop/esp-idf/components/wpa_supplicant/include/esp_supplicant -IC:/Users/user/Desktop/esp-idf/components/nvs_flash/include -IC:/Users/user/Desktop/esp-idf/components/pthread/include -IC:/Users/user/Desktop/esp-idf/components/perfmon/include -IC:/Users/user/Desktop/esp-idf/components/asio/asio/asio/include -IC:/Users/user/Desktop/esp-idf/components/asio/port/include -IC:/Users/user/Desktop/esp-idf/components/cbor/port/include -IC:/Users/user/Desktop/esp-idf/components/coap/port/include -IC:/Users/user/Desktop/esp-idf/components/coap/port/include/coap -IC:/Users/user/Desktop/esp-idf/components/coap/libcoap/include -IC:/Users/user/Desktop/esp-idf/components/coap/libcoap/include/coap2 -IC:/Users/user/Desktop/esp-idf/components/console -IC:/Users/user/Desktop/esp-idf/components/nghttp/port/include -IC:/Users/user/Desktop/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/Users/user/Desktop/esp-idf/components/esp-tls -IC:/Users/user/Desktop/esp-idf/components/esp_adc_cal/include -IC:/Users/user/Desktop/esp-idf/components/esp_gdbstub/include -IC:/Users/user/Desktop/esp-idf/components/esp_hid/include -IC:/Users/user/Desktop/esp-idf/components/tcp_transport/include -IC:/Users/user/Desktop/esp-idf/components/esp_http_client/include -IC:/Users/user/Desktop/esp-idf/components/esp_http_server/include -IC:/Users/user/Desktop/esp-idf/components/esp_https_ota/include -IC:/Users/user/Desktop/esp-idf/components/protobuf-c/protobuf-c -IC:/Users/user/Desktop/esp-idf/components/protocomm/include/common -IC:/Users/user/Desktop/esp-idf/components/protocomm/include/security -IC:/Users/user/Desktop/esp-idf/components/protocomm/include/transports -IC:/Users/user/Desktop/esp-idf/components/mdns/include -IC:/Users/user/Desktop/esp-idf/components/esp_local_ctrl/include -IC:/Users/user/Desktop/esp-idf/components/sdmmc/include -IC:/Users/user/Desktop/esp-idf/components/esp_serial_slave_link/include -IC:/Users/user/Desktop/esp-idf/components/esp_websocket_client/include -IC:/Users/user/Desktop/esp-idf/components/expat/expat/expat/lib -IC:/Users/user/Desktop/esp-idf/components/expat/port/include -IC:/Users/user/Desktop/esp-idf/components/wear_levelling/include -IC:/Users/user/Desktop/esp-idf/components/fatfs/diskio -IC:/Users/user/Desktop/esp-idf/components/fatfs/vfs -IC:/Users/user/Desktop/esp-idf/components/fatfs/src -IC:/Users/user/Desktop/esp-idf/components/freemodbus/common/include -IC:/Users/user/Desktop/esp-idf/components/idf_test/include -IC:/Users/user/Desktop/esp-idf/components/idf_test/include/esp32 -IC:/Users/user/Desktop/esp-idf/components/jsmn/include -IC:/Users/user/Desktop/esp-idf/components/json/cJSON -IC:/Users/user/Desktop/esp-idf/components/libsodium/libsodium/src/libsodium/include -IC:/Users/user/Desktop/esp-idf/components/libsodium/port_include -IC:/Users/user/Desktop/esp-idf/components/mqtt/esp-mqtt/include -IC:/Users/user/Desktop/esp-idf/components/openssl/include -IC:/Users/user/Desktop/esp-idf/components/spiffs/include -IC:/Users/user/Desktop/esp-idf/components/ulp/include -IC:/Users/user/Desktop/esp-idf/components/unity/include -IC:/Users/user/Desktop/esp-idf/components/unity/unity/src -IC:/Users/user/Desktop/esp-idf/components/wifi_provisioning/include -I../components/mongoose -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dirty\" -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/mqtt_server.c.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\mqtt_server.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/mqtt_server.c.obj   -c ../main/mqtt_server.c
In file included from ../main/mqtt_server.c:15:
../main/mqtt_server.c: In function '_mg_mqtt_dump':
../main/mqtt_server.c:94:11: error: implicit declaration of function 'pcTaskGetName'; did you mean 'pcTaskGetTaskName'? [-Werror=implicit-function-declaration]
  ESP_LOGI(pcTaskGetName(NULL),"%s=%x %x", tag, buf[0], buf[1]);
           ^~~~~~~~~~~~~
C:/Users/user/Desktop/esp-idf/components/log/include/esp_log.h:329:81: note: in definition of macro 'ESP_LOG_LEVEL'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                 ^~~
[... cut ...]
../main/mqtt_server.c:252:13: warning: passing argument 2 of 'esp_log_write' makes pointer from integer without a cast [-Wint-conversion]
    ESP_LOGD(pcTaskGetName(NULL), "c->fd=%p sub->c->fd=%p", c->fd, sub->c->fd);
             ^~~~~~~~~~~~~~~~~~~
C:/Users/user/Desktop/esp-idf/components/log/include/esp_log.h:331:81: note: in definition of macro 'ESP_LOG_LEVEL'
         else if (level==ESP_LOG_DEBUG )     { esp_log_write(ESP_LOG_DEBUG,      tag, LOG_FORMAT(D, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                 ^~~
C:/Users/user/Des

nopnop2002, 2021-04-22:
Can you build this ??

https://github.com/espressif/esp-idf/tree/master/examples/get-started/hello_world

lucagir, 2021-04-22:
Hello,
I tried and managed to complete the procedure

lucagir, 2021-04-22:
Hello again,
i solved the above problem.
As a newbie, I didn't have the latest version of the compiler of the IDF compiler.
By updating esp-idf I have overcome this above problem.

But sadly I ran into another problem in the final stage of the build.

how can i solve?

`.dir\main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj   -c ../main/main.c
../main/main.c: In function 'wifi_init_sta':
../main/main.c:242:2: error: implicit declaration of function 'dns_setserver'; did you mean 'dhcps_dns_setserver'? [-Werror=implicit-function-declaration]
  dns_setserver(0, &d);
  ^~~~~~~~~~~~~
  dhcps_dns_setserver
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

nopnop2002, 2021-04-22:
Try latest.

closed 2022-03-25 by nopnop2002