nopnop2002/esp-idf-mqtt-broker #34: Upload issues

41linea45, 2021-08-03:
Hello 

When I typed "idf.py flash monitor" i saw these issues. I can build this example. and I can build and upload Heelo world sample code.
I tried many times . Every time I download last version Mongose and Your code. Please see these pictures. What is my fault?

https://i.hizliresim.com/c5zr0w2.PNG
https://i.hizliresim.com/nu3upxb.PNG
https://i.hizliresim.com/qm2vx0y.PNG

nopnop2002, 2021-08-03:
```
git clone https://github.com/nopnop2002/esp-idf-mqtt-broker
cd esp-idf-mqtt-broker
mkdir -p components
cd components/
git clone https://github.com/cesanta/mongoose.git
cd mongoose/
echo "idf_component_register(SRCS \"mongoose.c\" INCLUDE_DIRS \".\")" > CMakeLists.txt
cd ../..
idf.py set-target esp32
idf.py menuconfig
idf.py flash monitor
```

nopnop2002, 2021-08-03:
Thank you for reporting.

Compliant with mongoose ver7.4.

41linea45, 2021-08-04:
Thank you for help. I downloaded code . I saw one more Warning pls look at pic.

https://www.hizliresim.com/bzdmvk3

41linea45, 2021-08-04:
Hello 

I created little mqtt system. 2 each esp32 mqttclient 1 each android application and connected esp32 mqtt broker client connects and publish message just one time . after that clients wifi lost connection . 

when i try esp32 clients to connect another mqtt broker or another mqtt es8266 , esp32 clients works properly.

So, I think that it may a failurw in mqtt broker. Mqtt broker cuts wifi client connection after receive message from esp32 clients or android client.

Please solve this problem . I need this Broker.

Best regards.

https://www.hizliresim.com/j2uwocz

nopnop2002, 2021-08-04:
Try changing the maximum station connection.

https://user-images.githubusercontent.com/6020549/110200317-a4d19d80-7ea0-11eb-84ec-21f78f97930b.jpg

41linea45, 2021-08-04:
Hello

I changed Maximum STA connection 10 each. But Result is same unfortunately.
when Every one message , wfi lost connection.

https://www.hizliresim.com/l32vi9j

nopnop2002, 2021-08-06:
If the MQTT client has a duplicate client ID, it will be disconnected.

Ex)
ESP32 connects with client ID = [CLIENT1].

ESP8266 connects with client ID = [CLIENT1].

ESP32 is disconnected.