nopnop2002/esp-idf-mqtt-broker #45: Client disconnects when subscribe to any topic

alegro96, 2023-12-18:
I can connect connect a client to the broker, but the client disconnect when I try to subscribe to a topic.

The broker gives the following messages:

I (455471) BROKER: MQTT_CMD_SUBSCRIBE
I (455471) BROKER: SUB ADD 0x37 []

What am I doing wrong?

nopnop2002, 2023-12-18:
Can you share your code or commands?

alegro96, 2023-12-18:
I used exactly your code. I configured the wifi-ssid and password in menuconfig and flashed the Code to my esp32.
Afterwords I tried to connect to the broker with postman. This worked well and I can publish messages.
But in the Moment I try to subscribe to a topic, postmen loses its connection to the broker.

nopnop2002, 2023-12-18:
This is my logging.
```
I (6715) SUBSCRIBE: CONNECTED to mqtt://192.168.10.167:1883
I (6725) SUBSCRIBE: SUBSCRIBED to #
I (6735) BROKER: MQTT_CMD_SUBSCRIBE
I (6735) BROKER: SUB ADD 0x38 [#]
```

>I (455471) BROKER: SUB ADD 0x37 []

This indicates that the topic to subscribe to is blank.

alegro96, 2023-12-19:
I used the internal MQTT-Subscriber via menuconfig and this one works. I get exactly the same log as you got:

I (5603) SUBSCRIBE: CONNECTED to mqtt://192.168.178.50:1883
I (5613) SUBSCRIBE: SUBSCRIBED to #
I (5623) BROKER: MQTT_CMD_SUBSCRIBE
I (5623) BROKER: SUB ADD 0x38 [#]

Seems like my postmen settings are incorrect. I tried to use MQTTX to connect and subscribe and get the same problem:

Connecting to the Broker works:

I (661763) BROKER: CONNECT
I (661763) BROKER: cid=[] willFlag=0

Subscribe a topic leads to disconnection:

I (715013) BROKER: MQTT_CMD_SUBSCRIBE
I (715013) BROKER: SUB ADD 0x39 []
I (715053) BROKER: MG_EV_CLOSE 0x39
I (715053) BROKER: free_size(MALLOC_CAP_8BIT):181224
I (715053) BROKER: free_size(MALLOC_CAP_32BIT):222280

It also seems like I try to subscribe to a blank topic. I think I have to figure out the correct settings.

alegro96, 2023-12-19:
Got it now.  I used the wrong MQTT-Version in postmen. I changed from V5 to V3 and everything works fine.

Thank you!

closed 2023-12-19 by alegro96