nopnop2002/esp-idf-mqtt-broker #48: What is the use of FAT file system in this project?

Bosemani, 2024-06-10:
Hai @nopnop2002 ,
    First, I'm appreciating your work. 
I'm having doubt, in your code main I found this part
`	// Initializing FAT file system 

	char *partition_label = "storage";
	wl_handle_t s_wl_handle = mountFATFS(partition_label, MOUNT_POINT);
	if (s_wl_handle < 0) {
		ESP_LOGE(TAG, "mountFATFS fail");
		while(1) { vTaskDelay(1); }
	}`

Why we need this storage partition?

Best Regards
Bose

nopnop2002, 2024-06-10:
As you pointed out, this is completely unnecessary code.

I never noticed it until now.

Initializing storage partitions is not required for this project.

Thank you for your report.

Bosemani, 2024-06-10:
Thank you for conformation.

closed 2024-06-10 by Bosemani