nopnop2002/esp-idf-mqtt-broker #49: is it possible to create username and password for clients?

jbsaibes, 2024-08-24:
i was wondering if it is possible to assign a username/password for clients that would connect to the broker?

nopnop2002, 2024-08-25:
This is a good idea.
Try latest.
[image]

jbsaibes, 2024-08-25:
yes i just tried and it works thanks! next request :) two nice features to have :
1) having a list of clients ( cid, user and password) that broker can use for authentication
2) having list of topics that above clients can subscribe/publish to

nopnop2002, 2024-08-25:
Are there any open source brokers with this kind of functionality?

jbsaibes, 2024-08-25:
yes there is https://github.com/eclipse/mosquitto it one of the most popular mqtt brokers

nopnop2002, 2024-08-25:
What settings do I need to do to achieve this?

having a list of clients ( cid, user and password) that broker can use for authentication
having list of topics that above clients can subscribe/publish to

jbsaibes, 2024-08-25:
https://mosquitto.org/man/mosquitto-conf-5.html check general options>acl file option  and "password_file" option .there is a description there.

nopnop2002, 2024-08-25:
Thank you for the information.
I'll try to investigate, but it will take some time.