nopnop2002/esp-idf-mqtt-broker #51: Suspected Memory Leak in Server event handler.

William-Irvine, 2024-09-17:
Hello Nopnop2002,

Sorry if this is not the case, but I believe there may be a small memory leak in fn() function in the file mqtt_server.c.

Similar to closed issue #35 where sub->topic.ptr was not released when the subscription was removed from the list after being set via the mg_strdup function on connection, I believe it is the same case for the will->payload.ptr.

I think that after line 456:

			    free((void *)will->topic.ptr);

the next line should be:

                        free((void *)will->payload.ptr);

as it is set on line 292 with:

                        will->payload = mg_strdup(payload);

Apologies if I am mistaken, but I do believe this is causing an issue after many disconnection and re-connections.

Great project!

Cheers,
Bill

nopnop2002, 2024-09-17:
Thank you for your report.
Your point is correct.
I fixed this issue.

William-Irvine, 2024-09-18:
No problem!

closed 2024-09-18 by nopnop2002